ManufacturedMotion/Hex #5: Header File with Configuration

Diacoba1, 2024-02-04:
Write code that takes in an offsets.h file and can use each of our offsets in the code. Should be flexible enough that we just need to specify whose bot it is, and the rest is abstracted away.

DillonKane, 2024-02-05:
First rev done. Could not dynamically create an array of axis objects and then use a for loop to define motor pins, etc. Instead that is hard coded for now -- this was what I was trying:

```
Axis axes[6][3]; 
  for (uint8_t i = 0; i < 6; i++) {
    for (uint8_t j = 0; j < 3; j++) {
        axes[i][j] = Axis(pwm_pins[i][j], min_pos[i][j] * scale_fact[i][j], max_pos[i][j] * scale_fact[i][j]);
    }
  }
```
This was not correctly meshing with the servos

Diacoba1, 2024-02-29:
- [ ] PWM pins defined per user
- [ ] Swap zero points for L1 & L2 for Danny and Zack
- [ ] Move `#define USER DILLON` to before the includes of the Arduino file

DillonKane, 2024-02-29:
https://github.com/ManufacturedMotion/HexapodController/pull/2

closed 2024-02-29 by DillonKane