ManufacturedMotion/Hex #145: Dead Code Cleanup

Iacobacci, 2024-11-24:
Ensure deprecated code is removed. Necessary after the Fall Surge

DillonKane, 2025-01-13:
I think we caught a lot of this during the first pass, but here is what I found this time around.

Undefined functions:

- [hexapod.hpp firstStepSetup](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/hexapod.hpp#L45)
- [leg.hpp linearMove](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/leg.hpp#L18)
- [leg.hpp runLegSpeed](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/leg.hpp#L23)
- [leg.hpp setLeg](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/leg.hpp#L28)
- [leg.hpp _linearMovement](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/leg.hpp#L48)
- [position.hpp addPos](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/position.hpp#L18)
- [position.hpp subtractPos](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/position.hpp#L21)
- [position.hpp distanceFromOrigin](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/position.hpp#L22)
- [three_by_matrices.hpp add](https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/real_time/teensy_code/lib/HexapodController/src/three_by_matrices.hpp#L29)

There is also this file I think we no longer need to track since we have our config.hpp with the same information
https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/offsets.md

@zschwid since you defined most of these functions can you give a quick look here and let me know which we will populate and which to remove? I can't remember if we discussed this in Nov and don't want to cut anything we will fill out

DillonKane, 2025-02-02:
https://github.com/ManufacturedMotion/Hexapod/pull/154

closed 2025-02-02 by DillonKane