ManufacturedMotion/Hex #157: Battery Voltage Measurements

Iacobacci, 2025-02-14:
- [ ] Have the battery voltage stored in JSON format
- [ ] Send the battery voltage to the pi every time the voltage changes a certain value
- [ ] On the pi side convert the battery voltage to a percentage -> likely using a lookup table

Flow:
Teensy has the voltage divided value on a pin -> Teensy converts the divided value into a usable measure and applies noise filtering -> Teensy sends this value in a JSON blob over serial -> "Teensy" ROS node on pi is connected to UART/Serial and picks up that value -> "Teensy" Node publishes that to a topic -->  node on the Pi is listening, picks up that value, and converts it to a percentage.

DillonKane, 2025-02-20:
to fix blocking issue we can use polling counters: think of embedded course. static var for counter and increment every loop instead of using a while loop.

DillonKane, 2025-02-22:
when working on pi side we want a soft warning at 3.5V per cell and hard cutoff at 3.3V

DillonKane, 2025-02-22:
first two bullets accomplished with these updates: https://github.com/ManufacturedMotion/Hexapod/pull/178

DillonKane, 2025-04-17:
lithium ion open cell voltage

closed 2025-06-29 by DillonKane