ManufacturedMotion/Hex #171: Strafing Twist -> Walk Setup Functions

Iacobacci, 2025-02-20:
Create a new package `gait` + node `teensy_gait` that takes in twist strafe values and converts it to the hexapod.walk_setup on the teensy.

- Subscribe to the /cmd_vel topic
- Convert this to the format needed for the teensy gait
- Publish the formatted command to the /to_teensy topic

Right Stick X-Axis: Linear +/- X (Holonomic Mode (Strafing) -- Walking in a direction without turning in that direction)
Right Stick Y-Axis: Linear +/- Y (Holonomic Mode (Strafing) -- Walking in a direction without turning in that direction)

Reference (but do better): mk2/code/non_real_time/container/src/ros2_ws/src/gait_node/gait_node/gait.py --  note: this is untested and may not work.

Existing Teensy Node (requires rework): https://github.com/ManufacturedMotion/Hexapod/blob/main/mk2/code/non_real_time/container/src/ros2_ws/src/teensy_node/teensy_node/uart_teensy.py

Change these: 

       ```
 self.publisher_ = self.create_publisher(String, '/teensy_receive', 10)
        self.subscriber = self.create_subscription(String,'teensy_send',self.send_to_teensy,10)
```

```
 self.publisher_ = self.create_publisher(String, '/from_teensy', 10)
 self.subscriber = self.create_subscription(String,'/to_teensy',self.send_to_teensy,10)
```

DillonKane, 2025-02-27:
```
ros2 run teensy_node teensy_node

ros2 run gait teensy_gait
```

DillonKane, 2025-03-02:
https://github.com/ManufacturedMotion/Hexapod/pull/186

closed 2025-03-02 by DillonKane