ManufacturedMotion/Hex #212: change DEBUG from DEBUG_LEVEL

DillonKane, 2025-03-23:
instead of 1 or 0 give more levels for debug printing

i.e.
0 - no debug
1 - print serial msgs
2 - print leg movement calculations, etc.

DillonKane, 2025-03-23:
make serial class with function that take msg and min log level.

ex: <class function>.log(<msg>, 3)
only logs the message if the DEBUG_LEVEL >= 3 in user_config

zschwid, 2025-04-13:
Use preprocessor to handle debug levels so that debug commands not in use get compiled out

DillonKane, 2025-04-22:
WIP: https://github.com/ManufacturedMotion/Hexapod/tree/serial_logging

I have most of the logging handled by the preprocessor on above branch but hit a snag. The current parser code for json/gcode/serial commands requires a pointer to our hexapod instance to be passed around. The only alternative to this I am thinking of now is to have the code to execute json/gcode commands live inside of Hexapod.cpp. I don't really like this though, I want to try and keep the parsing in separate source files

DillonKane, 2025-04-25:
remove serialHandler and just use serial.print

kill gcode support!!

DillonKane, 2025-05-04:
https://github.com/ManufacturedMotion/Hexapod/pull/231

closed 2025-05-04 by DillonKane