ManufacturedMotion/Hex #218: Update rpi to teensy command timing to match expected time to complete command

zschwid, 2025-03-30:
When RPi sends a command to teensy to move Xmm at Ymm/s, it should send the next command X/Y seconds later, or a bit before, depending on the speed of command parsing on the teensy. If the teensy takes N ms to receive and parse a command, then the RPi should provide the new commands at least N ms before the previous command is expected to finish.

zschwid, 2025-05-05:
Necessary to properly use new gait.

DillonKane, 2025-05-07:
https://github.com/ManufacturedMotion/Hexapod/tree/move_time_updates

this is looking good so far based on ros logs. Need to test on robot.

the speed scaling is a bit slow but I figure we can ramp it up once we have reprinted the new s2 for better strength

DillonKane, 2025-05-07:
https://github.com/ManufacturedMotion/Hexapod/pull/235

closed 2025-05-07 by DillonKane